Part-DB/Part-DB-legacy #5: Ändern der Kategoriezuweisung nicht möglich

jlegen, 2017-08-30:
Hallo,
bin auf der Suche nach einer einfachen Bauteileverwaltung auf part-db gestossen - scheint genau das Richtige für mich zu sein, einfach und intutiv, und sieht auch noch gut aus.
Installation war problemlos ("nextgen", PHP7.0.22), Kategorien schnell angelegt, erste Bauteile eingefügt, alles paletti.
Was nun allerdings nicht funktioniert: Ich möchte einige Bauteile von der "falschen" in die "richtige" Kategorie  verschieben. Es passiert einfach nichts, ausser dass im rechten "suchen/hinzufügen"-Feld die **neue** Kategorie-ID erscheint in der Form "__ID__=66". Bauteil bleibt aber in der alten Katgeorie hängen.
In der Online-Demo allerdings scheint das zu funktionieren. Das PHP-Log gibt bei mir wohl erstmal nicht viel her, da es zwar bei jeder Aktion einige Warnings gibt, beim Kategoriezuweisen aber keine besonderen/anderen.

Dabei ist an dieser Stelle generell etwas unschön, dass der "Bauteil ändern"-Dialog keinerlei Feedback liefert, ob "Änderungen übernehmen" überhaupt etwas macht/gemacht hat. Dito "verwerfen" - kann man klicken, unabhängig davon, ob es überhaupt Änderungen gibt, bzw. ohne das klar wird, ob damit auch Änderungen gemeint sind, die man eben "übernommen" hat etc.
Gerade bei "Tablet-Bedienung" ist das fehlende Speichern-Feedback etwas irritierend.

jbtronics, 2017-08-30:
Was für eine Version benutzt du denn?
Eine der Versionen aus den Releases? Denn die sind schon recht alt...
Am neusten ist immer die Version aus dem nextgen Branch, da sollte der Fehler eigentlich nicht mehr auftreten...

jlegen, 2017-08-30:
"nextgen"-Download aus "nextgen"-Branch - Stand gestern. 
Screenshot:

[image]

jbtronics, 2017-08-30:
Was für einen Browser benutzt du? Hast du Javascript aktiviert?

Kannst du dir mal die neuste Version ziehen, weil ich hab gestern ein Problem mit diesen Feldern behoben gehabt, möglicherweise hast du deine Version geladen gehabt bevor ich es gefixt hatte.

jlegen, 2017-08-30:
Bingo, das Update hat funktioniert, vielen Dank!
Man kann zwar nicht "sehen", dass bei "Änderungen übernehmen" etwas übernommen wird, aber das erneute Aufrufen des Bauteils zeigt dann die geänderte Kategorie...

jbtronics, 2017-08-30:
Normalerweise sollte es immer gespeichert werden, wenn man auf Übernehmen klickt, wenn nicht sollte eine Fehlermeldung erscheinen.

Ich kann aber gerne die Tage noch eine Option hinzufügen, dass er eine Nachricht anzeigt, die bestätigt das gespeichert wurde.

jlegen, 2017-08-30:
Normalerweise ist das auch sicherlich der Fall - man kann sich (insbesondere mit Tablet) aber erst sicher sein, wenn man sich die geänderten Daten wieder anschaut.
Statt einer Nachricht "erfolgreich gespeichert" wäre es aus meiner Sicht sogar besser (und einfacher?), nach dem Klicken von "Änderungen übernehmen" wieder die Ansicht davor, also "Detailinfo zu ..." mit dem Button unten "Angaben verändern" anzuzeigen! 
Ich glaube, das entspräche auch etwa dem Verhalten, "wie man's gewohnt ist"... :-)

jbtronics, 2017-08-30:
Danke für die Anregungen.
Ich werde mich die Tage mal versuchen, es umzusetzen.

jlegen, 2017-08-30:
Das wäre cool!
Der aus Nutzersicht zwangsläufig "steinige Beginn" besteht ja aus "mindestens hunderten" wenn nicht "unzähligen" Bauteileingaben und ggf. Änderungen (bis man sich über Kategorien im Klaren ist). Da hilft m.E. jede kleine Optimierung des "Workflows"...
Bauteileingabe ist übrigens "verwandt" mit meiner Edit-Anregung: Legt man ein Bauteil an (= klickt auf "Bauteil erstellen"), würde ich auch eher die "Detailansicht" des neuen Bauteils erwarten, und nicht den "Edit"-Modus (den ich eigentlich dachte, mit "Bauteil erstellen" gerade verlassen zu haben).

Wenn es dann noch eine schnelle Möglichkeit gäbe, sofort ein weiteres Bauteil derselben Kategorie eingeben zu können, wäre das optimal :) - wenn man aktuell links nochmal auf die gerade aktive Kategorie klickt, landet man ja nicht (wie man vielleicht erwarten würde) wieder in der entsprechenden Bauteilliste, sondern man "de-selektiert" die Kategorie irgendwie optisch (was bei einer Navigationsleiste imo nicht wirklich zu erwarten wäre). 
D.h. derzeit klicke ich dann beherzt zweimal auf die Kategorie, und erhalte dann wieder die Liste mit den Bauteilen, um oben dann "neues Bauteil anlegen" zu wählen.

jbtronics, 2017-08-30:
Es gibt sogar schon eine Möglichkeit ein neues Teil in der Kategorie anzulegen :) : Du musst auf der Bauteile Bearbeitungsseite ganz nach unten scrollen und dann auf Neues Bauteil erfassen. Das neue Teil hat dann die selbe Kategorie wie das alte.

Ich werde das nachher Mal ins Wiki schreiben, dann findet man das vermutlich besser.

jlegen, 2017-08-30:
Ah, tatsächlich - in blau hätte ich ihn evtl. auch gefunden... ;-)

jbtronics, 2017-09-01:
Ich hab jetzt mal die Möglichkeit eingebaut direkt auf die Übersichtsseite umzuleiten nachdem ein Teil angelegt/gespeichert wurde. Aus Gründen der Kompatibilität zum alten Verhalten, ist es zwar standardmäßig deaktiviert, man kann es aber einfach mit den Haken unter Funktionen -> Bauteilebearbeitung in den Einstellungen aktivieren.
Außerdem kann man das Verhalten einmalig ändern, in dem man auf den Button zum anlegen/speicher, einen Rechtsklick macht. Dann wird das Gegenteil des standardmäßigen Verhaltens ausgeführt.

Die Verbesserungen findest du in der aktuellen Version aus dem nextgen Branch.

closed 2017-09-02 by jbtronics

jlegen, 2017-09-03:
vielen Dank, funktioniert prima!
Das Sahnehäubchen wäre allerdings, wenn es auch im neuen "Modus" möglich wäre, nach Neuanlegen/Ändern direkt ein weiteres neues Bauteil anlegen zu können :-)
(also "Neues Bauteil erfassen" unten auf der Seite)

jbtronics, 2017-09-03:
@jlegen Ok, ich schaue Mal was ich tun kann.

jlegen, 2017-09-04:
Vielen Dank! Damit ist die Dateneingabe richtig rund, finde ich...
Zu den "Baugruppen" hätte ich auch noch eine Idee - da mach' ich aber besser einen neues Ticket auf... :-)

jbtronics, 2017-09-04:
Ja, ist ein neues Ticket wäre ganz gut, dann behält man den Überblick besser