Part-DB/Part-DB-legacy #6: Einfachere Baugruppen-Handhabung

jlegen, 2017-09-05:
Das "Baugruppen"-Konzept finde ich grundsätzlich sehr gut - so kann man den Überblick über den Bedarf von Projekten behalten, zumindest wäre das meine Anwendung :-)

Allerdings - habe ich eine Baugruppe angelegt/aufgerufen, und möchte nun Bauteile zuordnen, bin ich etwas ratlos. Es gibt eine "Suchzeile", mit der ich das Bauteil finden/listen kann - aber das hilft mir nur bedingt. Ich möchte z.B. ein LCD hinzufügen, kenne die Bezeichnung (meiner "vorrätigen") aber nicht auswendig. Einfach nach "Display" oder "LCD" (an die ich mich als Kategorien-Name erinnern würde) helfen nicht weiter, da die Kategorie offenbar nicht zur Suche gehört. "Alle Teile" listen ist ab ca. 1000 oder so nicht praktisch :)

* Idee 1: bei der Suche auch die "matchende" Kategorie listen
* Idee 2: idealerweise würde ich für die Zuordnung eines Bauteils zu einem Projekt einfach die bereits vorhandene Kategorien-Baumstruktur nutzen wollen. Z.B. indem ich beim normalen Auflisten von Bauteilen und in deren Detailansicht ein "Knöpfchen" zum Zuordnen zu einem Projekt hätte.
Die möglichen Projekte könnten evtl. in der Form eines "Popups" zur Auswahl gestellt werden, mit dem jeweils zuletzt angesprochenen "ganz oben".
Oder aber man versetzt ein Projekt "global" in einen Hinzufügemodus, und wählt Bauteile ähnlich wie bei einem Warenkorb aus. Wobei das vermutlich eher aufwendiger und etwas intransparenter wäre...

jbtronics, 2017-09-25:
Sry, dass es etwas länger gedauert hat, ich war ziemlich mit dem Benutzersystem beschäftigt.

Dafür habe ich jetzt deine Ideen alle mehr oder weniger umgesetzt:
1. Wenn man auf der Baugruppenseite nach Teilen zum Hinzufügen sucht, dann sucht er in allen Feldern des Bauteils (außer Lieferantenname)
2. Auf der Detailseite jedes Bauteils gibt es jetzt ein ausklappbares Panel, was anzeigt in welchen Baugruppen, dass Bauteil momentan verwendet wird. Ganz unten gibt es auch die Option, das Bauteil einer neu einer Baugruppe hinzuzufügen. Dabei ist die vorgewählte Baugruppe, jene, dessen Radiobutton auf der Baugruppenübersichtsseite ausgewählt ist.

Hoffe dass ist in etwa so wie du es dir vorgestellt hast.

jlegen, 2017-09-25:
Super, vielen Dank - werd' ich gleich morgen dann mal updaten und ausprobieren!

jlegen, 2017-09-26:
Ja, so klappt das nun hervorragend, danke! 👍