Part-DB/Part-DB-legacy #11: Buglet: Datenbank-Inkonsistenz durch Dateneingabe

jlegen, 2017-09-25:
Partdb Version vom 25.9.
Beim Hinzufügen eines Bildes zu einem neu erstellten Bauteil ist ein Bild zweimal hinzugefügt worden, was 'bootstrap-typisch' bei Tabletbenutzung häufiger passiert.
Danach einfach eines der Bilder wieder gelöscht - damit war dann allerdings die betreffende Kategorie nicht mehr aufrufbar. 
Fehlermeldung: sinngemäss 'Attachment Nr. n nicht vorhanden' (mit Button zum Neuladen, was hier natürlich nicht hilft).
In der DB war attachement Nr. n tatsächlich 'ordnungsgemäss' gelöscht, das entsprechende Bild mit Nr. n-1 vorhanden. Im Bauteil (parts) allerdings war als master-bild Nr. n eingetragen.
Manuelles Ändern der Tabelle parts auf 'n-1' hat das Problem behoben.
Ich hoffe, das war halbwegs verständlich... :)

jbtronics, 2017-09-25:
Ok,

war ein Problem in Folge des Benutzersystem, dass der master_picture Wert nicht null werden konnte.
Sollte jetzt gefixt sein.

Wenn du es bestätigen kannst, dann schließe ich diesen Issue.

jlegen, 2017-09-26:
Ja, sieht gut aus, vielen Dank!

jbtronics, 2017-09-26:
Fixed

closed 2017-09-26 by jbtronics