Part-DB/Part-DB-legacy #12: Bug: Anzahl vorhandener Bauteile wird auf 0 gesetzt

jlegen, 2017-09-27:
Hi,
kleiner Bug - die Anzahl der Bauteile ("Vorhanden") wird "unerwartet" auf 0 gesetzt, wenn
* man ein Bild hinzufügt; das Bauteilformular enthält alle Daten, aber angezeigte Anzahl ist  dann immer "0" - wirkt sich natürlich nur aus, fall man nochmal "Änderungen übernehmen" sagt, was allersdings schonmal vorkommen kann...
* wenn man ein Bauteil in eine andere Kategorie verschiebt, wird der Bestand ebenfalls auf 0 gesetzt - wobei man in diesem Fall den Kategoriewechsel ja immer per "Änderungen übernehmen" speichern muss, sodass die 0 dann gleich mitgespeichert wird...

jbtronics, 2017-09-27:
Ok, sollte jetzt gefixt sein.
War ein Problem, dass durch das Javascript entstand, dass sich um die Aktivierung/Deaktivierung des Bestandfeldes kümmert, wenn die "Bestand unbekannt" Checkbox geändert wird.

jlegen, 2017-09-27:
Danke, klappt nun!

closed 2017-09-27 by jbtronics