Part-DB/Part-DB-legacy #13: Feld für Kommentare/Notizen/URLs für 'Baugruppen'

jlegen, 2017-10-12:
...würde diese Funktion erheblich aufwerten, imo. Z.B. könnte man Referenzen auf den zugehörigen Code hinzufügen, oder Versionsinformationen, oder zus. Doku abseits von "Datenblatt"...

jbtronics, 2017-10-12:
Mittlerweile kann man ja schon Dateianhänge hinzufügen... 
Aber ich hatte bald vor, die Datenbankstruktur zu erweitern und unter anderem Kommentarfelder für Footprints/Kategorien (außerdem RoHS-Eigenschaft und Favoritenmarkierung) für Bauteile und, etc. einzubauen. Dann kann ich auch ein Kommentarfeld für Baugruppen, etc. einbauen.

jbtronics, 2017-10-21:
So in der aktuellen Version kann man allen Datenstrukturen, wie Footprints, Lagerorte, etc. und natürlich auch Baugruppen Kommentare hinzufügen.
Dies ist bei den Bauteilen z.B. unter Bearbeiten->Baugruppen möglich. Wenn man dort etwas hineingeschrieben hat, wird dieser Komentar auch in einem eigenem Panel auf der Baugruppenseite angezeigt (dort auch mit geparsten BBCode).

jlegen, 2017-10-23:
Supi, vielen Dank! Kleine Winzigkeit: Ist ein Kommentar mal eingetragen, kann man ihn auch unmittelbar aus der Baugruppenansicht editieren. 
Um diesen initial einzutragen, muss man allderdings in die Verwaltung nach 'Baugruppen anlegen' wechseln. 
Vielleicht liesse sich ja noch ein 'Kommentar bearbeiten' bei den Baugruppen selbst anzeigen, auch wenn's noch keinen gibt...

jbtronics, 2017-10-26:
Ok, habe das jetzt in der neusten Version eingebaut.

closed 2017-10-26 by jbtronics