Part-DB/Part-DB-legacy #18: Datenblattlinks in der Detailansicht

jlegen, 2017-11-19:
Während man in den Listen der "Übersichten" Links auf Datenblätter findet, fehlen diese in der "Detailansicht" eines Bauteils. Manchmal wäre es imo ganz praktisch, auch dort die Links zu finden.
Zumal es m.E. eigentlich üblicher Logik folgt, wenn an dieser Stelle alle Daten zu einem Bauteil zu finden sind...

jbtronics, 2017-11-19:
Ja macht Sinn, ich werde mal schauen wie man das umsetzen kann.

jbtronics, 2017-11-22:
So in der aktuellen nextgen Version gibt es jetzt auf der Übersichtsseite ein Dropdown, wo man auf die Datenblattlinks zugreifen kann.

Vermutlich kommt die Änderung auch noch in die Version 0.4.2, da es eigentlich nur eine kosmetische Erweiterung eines bestehenden Features ist.

jlegen, 2017-11-24:
prima, sieht gut aus, danke!
noch 2 winzige Kleinigkeiten, für die kein eigenes Ticket lohnt:
- der Link "Hilfe" links unten im Hauptmenü tut nix unter aktuellem Firefox, unter Chrome verweist er auf Github
- am besten bei Gelegenheit mal einen sed /obsolent/obsolet/ durchlaufen lassen :)

jbtronics, 2017-11-25:
Ok, ist in der Version 0.4.2 oder der aktuellen nextgen Version behoben.

closed 2017-11-25 by jbtronics