Part-DB/Part-DB-legacy #19: Eingabehilfe

jlegen, 2017-12-11:
...vielleicht habe ich ja nur die falsche Tastatur, aber derzeit scheint es mir nur mit Hilfe von Copy&Paste möglich zu sein, z.B. ein Ω -Zeichen eingeben zu können. Evtl. liesse sich eine Funktion der Art "Sonderzeichen einfügen" realisieren? 
Finden sich bestimmt noch ein paar andere nützliche Symbole...

mfechner, 2017-12-11:
Hi, es ist eigentlich üblich R, K, M, G zu verwenden.
Für 3,5Ω schreibt man normalerweise 3R5, usw.

jlegen, 2017-12-11:
Stimmt, aber eigentlich sollte das nur ein Beispiel sein; z.B. dürfte 220μF (22μH)  durchaus gebräuchlich sein - musste ich mir hier auch per C&P einfügen, da mein Tablet kein ‚AltGr‘ hat... ;-)
Würde evtl. auch gerne mal notieren, dass etwas 13£ gekostet hat, oder 22€...
Ist letztlich aber ja nur ein kleines Goodie, nicht wirklich wichtig...

jbtronics, 2017-12-11:
Ich find die Idee aber ganz gut. Es gibt scheinbar Bibliotheken für virtuelle Tastaturen, daher sollte es nicht so schwer sein, das einzubauen. Das könnte auch helfen, wenn man ein System ohne Tastatur hat...
Es wird dann vermutlich in den Textfeldern ein Button geben mit dem man das virtuelle Keyboard aktivieren kann, und dann auch direkt griechische Zeichnen einfügen.

Zu den 13£ und 22€, so mittelfristig will ich die Unterstützung von verschiedenen Währungen einbauen, dann dürfte das gut über die Preisfunktion der Teile funktionieren.

Ich werde die Tage Mal probieren die Tastatur einzubauen...

jbtronics, 2017-12-15:
So in der aktuellen "nextgen" Version findet sich eine Funktion, dass man verschiedene Sonderzeichen über die normale Tastatur eingeben kann, indem man ALT + eine bestimmte Taste drückt. Dies sollte an allen Inputs mit Texteingabe funktionieren.
Eine Übersicht über die bisher implementierten Zeichen findet man hier: https://raw.githubusercontent.com/jbtronics/Part-DB/nextgen/documentation/special_chars-layout.png

Ich lasse den Issue noch ein mal offen, vielleich baue ich später noch eine Bildschirmtastatur ein.