Part-DB/Part-DB-legacy #21: Error on first install

GeorgianBorca, 2017-12-16:
Hello, I get an erro in browser when trying to access the install.php
Could you please help me?
Here is a picture with the error.
I'm using a raspberry pi as server.
[image]

jbtronics, 2017-12-16:
This look like an permission error. Can you try to run the commands from this guide (https://github.com/jbtronics/Part-DB/wiki/EN%3A-Installation#set-access-rights-for-directories).

When this don't work, try to run chmod 755 on the Part-DB main folder.

GeorgianBorca, 2017-12-16:
sudo chmod 755 Part-DB-0.4.2 solved the issue. 
Now i have a database problem. 
[image]

jbtronics, 2017-12-16:
@GeorgianBorca Jan Böhmer:
Does it work if you uncheck the checkbox "insert space into pdo string"? Is the password correct? 

Also you could try to create a new user for part-db, which has full access to the database. (Furthermore it is much safer to not use the MySQL root user for that)

jbtronics, 2017-12-16:
@jbtronics Part-DB can not create a database for you, so you have to create the Database LAGER manually with a tool like phpmyadmin. Check the wiki on GitHub if you need help with this.

GeorgianBorca, 2017-12-16:
Thank you for your help. I have installed part-db on my server (hosted by easyname)  and it's working. I'm trying again a clean install on my raspberry pi to fix the issue.

GeorgianBorca, 2017-12-16:
After creating a databese i entered this command "GRANT ALL ON `DATABASE NAME`.* TO 'phpmyadmin'@localhost IDENTIFIED BY 'PASSWORD';" in webbrowser logged in as root and i can no longer login to mysql. I tried with user root and with user phpmyadmin and i alwey get an error.

Console error:
pi@raspberrypi:/var/www/html/Part-DB-0.4.2 $ sudo mysql -u root -p
Enter password:
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)
Web browser error:
[image]

jbtronics, 2017-12-16:
You have to use the user "phpmyadmin" (without quotes) and the password you set in the last step for login in phpmyadmin, not root.

GeorgianBorca, 2017-12-16:
I did and i tried again to create a database and error again. 
[image]

GeorgianBorca, 2017-12-16:
console error:
[image]

jbtronics, 2017-12-16:
You have to login into a root MySQL shell via the commandline and run the command 

> GRANT ALL PRIVILEGES ON *.* TO 'phpmyadmin';

Then you can login into phpmyadmin with the phpmyadmin user and create a table and a user for Part-DB.

GeorgianBorca, 2017-12-16:
I really apreciete your help but I seem to be to dumb for this.
For some reason i can no longer login with the username "root" and the password for "root" account. 

using this set of commands 

> sudo apt-get install apache2 php7.0 mysql-server php7.0-mysql php7.0-curl libapache2-mod-php7.0 phpmyadmin php7.0-opcache php7.0-gettext

I set up a password for the user "phpmyadmin".  I was not asked to setup a password for "root acount".
Following some instructions from this website [https://superuser.com/questions/603026/mysql-how-to-fix-access-denied-for-user-rootlocalhost](url) I set up a password for the user "root".

After this i loged in using a web broswer with the user name "root" and the password i set up in the last step. I issued the `"CREATE DATABASE `LAGER`;" command and after i issued the "GRANT ALL ON `LAGER`.* TO 'phpmyadmin'@localhost IDENTIFIED BY 'PASSWORD';"` command. After this command, I was atomaticly loged out from web browser and unable to login again using the "root" username. I can login using the "phpmyadmin" user name but as described in the last pictures, i'm unable to create a database. I hope you understand my problems and can help. Thank you.

jbtronics, 2017-12-16:
Try to reset your mysql root password  (https://www.howtoforge.com/reset-forgotten-mysql-root-password). 
I think you has to stop your mysql-server with `sudo service mysql stop` and start it with `sudo service mysql start` after you reset the password.
Then you should be able to login with the root user into phpmyadmin and create a database for Part-DB

GeorgianBorca, 2017-12-16:
Here is a result:
[image]

jbtronics, 2017-12-16:
Try to run the mysql command in another terminal after running mysqld_safe and dont press Ctrl-C.

GeorgianBorca, 2017-12-16:
It worked using another terminal windows. I could reset the password. Thank you very much for your time.

GeorgianBorca, 2017-12-16:
I got it up and running but I can't login because didn't setup n user name.  I only set up an password for administrator account.
[image]

jbtronics, 2017-12-16:
The default login, is "admin" as username and "admin" as password (both without quotes).  They have (yet) nothing to do with the administrative password you set during installation.

After login in with default credentials you are asked to change the password for the admin user.

jbtronics, 2017-12-17:
Is it working now?

GeorgianBorca, 2017-12-17:
Yes, I got it all up and working. A raspberry 2 is perfect for the job. Thank you for your support.

closed 2017-12-17 by jbtronics