Part-DB/Part-DB-legacy #22: Bildformate

jlegen, 2017-12-28:
Kleine Unschönheit: Gibt man z.B. für einen Footprint ein Bild im JPG-Format an, so wird das zwar angenommen. hochgeladen und vermutlich irgendwo abgelegt - es wird aber "kommentarlos" nicht angezeigt.
Offenbar muss man nach PNG wandeln, damit es auch mit der Anzeige klappt (?).
Entweder sollte daher vielleicht bei der Eingabe/Upload eine (Fehler-)Meldung kommen, oder aber JPG könnten vielleicht auch angezeigt werden? :)

jbtronics, 2017-12-28:
Das ist interessant, denn eigentlich sollte sich der Browser um die bild Darstellung kümmern, und daher alle formate unterstützt werden, die der Browser kann...

Vermutlich wird die Datei nicht korrekt als Bild erkannt.

jbtronics, 2017-12-28:
Hat das Bild die Endung .jpeg oder .jpg und funktioniert es wenn du die Endung veränderst?

Nur um sicher zu gehen, du meinst die Art von Bild die Mann über die edit_footprint.php hochladen kann und keinen Anhang für ein Bauteil?

Ich werde mir das Mal genauer ansehen, denn ich sehe keinen Grund das es in Zukunft nicht auch mit JPEG Dateien (und gif, svg, etc) funktionieren sollte.

jlegen, 2017-12-29:
Ja, aufegefallen ist mir das bei edit_footprint.php. Dachte ich - ich glaube, ich habe mich da aber nur von der UI verwirren lassen; wenn man ein Bild hochlädt, passiert irgendwie "nichts", ausser dass der "Remove"-Button und etwas white space unterhalb erscheinen. Ich war davon ausgegangen, dass da dann die Bildvorschau erscheinen *sollte*... (auch das Eingabefeld bleibt leer bzw. auf Default).
Erst nach "Footprint anlegen" wird die Grafik sichtbar - allerdings auch bei *.jpg, wie's scheint... 
Also wohl eher ein Fehlalarm, sorry.

jbtronics, 2017-12-30:
Ok, dann schließe ich den Issue jetzt...

closed 2017-12-30 by jbtronics