Part-DB/Part-DB-legacy #26: Multiple labels (batch processing)

julianobst, 2018-07-07:
Are there any plans for a function to create multiple labels at once (e.g. all parts of a location or from part id X to Y) ?
Next step could be the export of the set as PDF file with multiple pages, CSV file (barcode as number) or as html page. The ultimate solution would be the integration of label sheets like Avery or Herma.

jbtronics, 2018-07-07:
One of my big goals for the future is a much better export (and import) system. 
When this works, it should be possible to implement batch Export for the Labels too...

I use a Label printer, which print one Label at once, so i did not needed Label Sheets yet. But i will try to implement this Feature for the future.

jbtronics, 2023-07-19:
This is available in the Part-DB rewrite: https://github.com/Part-DB/Part-DB-server

closed 2023-07-19 by jbtronics