Part-DB/Part-DB-legacy #27: Install is half english half german

whinis, 2018-08-09:
After installing and walking through the installer entirely in german using google translate I get to the final screen and even with english selected as the language I get some menus in english but most are in german.

jbtronics, 2018-08-09:
Has you php-gettext installed? 
In the Source Code the Strings are unfortunatly in german, and are translated to English via gettext, so If No gettext is available, all Texts are shown in German...

Could you send me your config.php from the Data/ folder (please remove your Database credentials from it)? 

Maybe the language setting was not saved, or a wrong locale was written into config...

whinis, 2018-08-09:
I managed to get it working, the installation instructions on the wiki are slightly confusing. It says to download the latest which is fair but then gives a command for another repo with what I assume is a highly outdated version.

Next it ask to set file permissions however in doing so the next command for composer will not work because write permission was taken away.

Finally while the text says to set permissions on documentation the commands given do not.

whinis, 2018-08-09:
Another problem I encountered is after setting permissions templates_c is not created nor can be created

Also both with 0.4.6 and 0.4.5 the sidebar which would show the categories and general navigation doesn't work and nothing is shown beyond the head classes such as Management, Devices, and Categories with no children

whinis, 2018-08-10:
After much tinkering the final issue was with my php configuration, Issue fixed minus the documentation

closed 2018-08-10 by whinis