Part-DB/Part-DB-legacy #29: Kommentar als Popup beim Mouseover

MrSmallpart, 2018-08-25:
Hallo,
ich habe bei meiner kleinen Datenbank ein kleines/großes Problem. Wenn ich irgend ein Bauteil suche und es dann aufrufen will, klicke ich ja auf dea betreffende Bauteil, bzw. auf den Namen (neben dem Bild). Bevor ich aber zum Klicken komme, kommt immer der Kommentar als Popup, wenn ich mit der Maus darüber fahre. Soweit, so schlecht (weil im Popup alle Zeichen vom html-Code auch sichtbar sind). Aber wenn ich nun da drauf klicke, komme ich zu dem Bauteil, bzw den Details. Soweit ist es dann wieder gut.  NUR dieses Popup vom Mouseover verschwindet nicht, sondern bleibt an der Stelle wie vor dem Aufrufen der Details. Und wenn ich ein anderes Bauteil aufrufe und noch eines, dann werden es immer mehr Popup´s.
Wenn ich das "weg" haben will, muß ich bei der Ansicht der Details einmal die Seite neu laden, dann ist es weg.  Aber deim Aufrufen des nächsten Bauteils fängt das Spiel wieder von vorne an.
Und das ist wirklich nervig.
Kann ich das irgendwo in den Einstellungen ändern?
Und wenn ich schon mal frage: Kann ich das Layout der Barcode-Labels irgendwo ändern? 

Beste Grüße
MrSmallPart
[image]
[image]

jbtronics, 2018-08-25:
Was für einen Browser benutzt du denn? Und welche Part-DB Version? Eine 0.4.x oder eine der 0.5-dev Versionen?

Zu den Barcode labeln: Die größe, Inhalt und die Formatierung (mit HTML) der Barcodes kannst du auf der Seite wo man die Barcodes erzeugen kann, ändern (zumindest in den aktuellen Versionen)... Oder was genau meinst du mit Layout?

jlegen, 2018-08-26:
Das mit den „persistenten“ Tiptools (Kommentaren im Mouse-over) kann ich bestätigen für 0.5-dev und z.B. Safari-Browser (ios).
Das mit dem „HTML-Code“ liegt aber wohl eher an der 1:1-kopierten Ali-Express-Prosa... :-)

MrSmallpart, 2018-08-27:
Hallo,
danke für die Antworten.  Browser benutze ich verschiedene, also Firefox, Google Chrome, oder auch Opera probiert. Bei allen kommt dieses Phänomen. Und nicht nur bei mir, sondern einem befreundeten Bekannten, der von seinem Browser auf meine DB zugreift.

@jlegen  die Vermutung bezüglich AliExpress kann ich nicht bestätigen. Habe eben einen "paar Worte Kommentar" geschrieben, ohne Html-Code, aber auch dieser bleibt konstant, selbst wenn ich die "Home"-Seite aufrufe. Erst, wie schon erwähnt, ich die Seite "refresche", geht der Kommentag weg.

[image]
[image]

jbtronics, 2018-08-29:
Bei der aktuellen Version aus dem master-Branch sollte das Problem mit den Tooltips endlich behoben sein...

Außerdem habe ich noch etwas eingebaut, dass er allen BBCode aus den Kommentaren entfernt, wenn er ihn als Tooltip zeigt, es sollte also kein HTML-Code Müll mehr in den Tooltips stehen...

jlegen, 2018-08-29:
bei mir unter Win10/Chrome klappt es nun "meistens" - also nicht immer... 
[image]

jbtronics, 2018-08-29:
Hast du Mal testhalber mit Ctrl-Shift-R neugeladen? Vielleicht hat er noch eine gecachte Skriptversion benutzt...

Das ist insgesamt ein ziemlich hartnäckiges Problem, ich kämpfe damit seit ich die BS tooltips implementiert habe....

jlegen, 2018-08-29:
ja, ich habe sogar explizit den Browsercache vorher gelöscht...

jbtronics, 2018-08-29:
Okay dann wird das damit zusammen hängen, dass das löschen der tooltips asynchron ist. Und dann nicht alle gelöscht werden, bevor neue erzeugt werden... Ich schaue Mal was sich da machen lässt...

jlegen, 2018-08-29:
ich hab' da keine Ahnung von, aber sowas könnte es nicht sein?

> [https://stackoverflow.com/questions/33584392/bootstraps-tooltip-doesnt-disappear-after-button-click-mouseleave]

MrSmallpart, 2018-08-30:
Es ist sehr informativ, dass alles zu lesen. Nur manches ist mir nicht so klar und geläufig, wie es hier beschrieben wird, da ich mich zwar mit elektronischen Bauteilen halbwegs auskenne, mit html und anderem Code nicht so sehr    ;-) .
Vielleicht sollte ich ein Update abwarten, wenn es irgendwie doch möglich ist, das Problem in den Griff zu bekommen.
Aber vielleicht darf ich noch eine andere Frage stellen:
Diese beschriebene DB habe ich auf einem Server. Eine "alte" PartDB 0.3.0 habe ich auf meinem PC über XAMPP laufen.  Ist es möglich, die Bauteile in die "neue" PartDB Version: 0.5.0-dev (development)
"NextGen" hinüber zu bekommen?
Ich habe schon soviel Internetseiten besucht, aber leider keine deutsche "Anleitung" bzw. Beschreibung gefunden und im Englischen bin ich nicht so gut. ;-(

Würd mich sehr freuen, wenn mir da jemand einen guten Tipp geben könnte.
Gruß
MrSmallpart

jbtronics, 2018-08-30:
@MrSmallPart im Prinzip reicht es die 0.5er Version in einen Ordner in XAMMP, zu kopieren (+ die Composer installieren oder zu entpacken) und dann den Ordner Data/ aus der alten Part-DB Version in die neue zu kopieren... Wenn du die neue Version dann im Browser aufrufst, dann wird er dich fragen die Datenbankstruktur zu aktualisieren, wenn du dem zustimmst, dann solltest du danach das neue Part-DB mit den alten teilen haben.. 

Ich würde aber dringend raten vorher ein Backup der Datenbank anzulegen (das geht zum Beispiel über phpmyadmin in XAMPP über exportieren)

MrSmallpart, 2018-08-30:
@ jbtronics  
Danke, ich werde es so mal ausprobieren.

MrSmallpart, 2018-08-30:
Jetzt habe ich das Passwort-Problem (hatte ich schon mal).
Ich habe testweise eine neue PartDB 050 angelegt, dann wie du schreibst, den alten Data-Ordner in den neuen kopiert und es wurde ein Update gemacht (von ver15 auf ver23. Dann will er, das ich das Passwort ändere. Jetzt bin ich mir nicht sicher, das von der "Neuen DB" oder der "Alten DB". Jedenfalls klappt es nicht, kommt immer die Fehlermeldung Passwort ist falsch.
Das Passwort von der neuen DB habe ich ja ganz neu vergeben (also kenne ich es) und das Password der alten DB weiß ich leider nicht mehr,, weil irgendwann vergessen und auch nicht gebraucht.
Wo kann ich das Passwort-Problem lösen?
Steht es in irgendeiner Config.php?

[image]

jbtronics, 2018-08-30:
@MrSmallpart schau Mal hier: https://github.com/Part-DB/Part-DB/wiki/Passwort-zurücksetzen

MrSmallpart, 2018-08-30:
ok, nur in der Vers030 hab ich die "reset_admin_pw.php" nicht. Kann ich die aus der Vers050 dafür nehmen?

jbtronics, 2018-08-30:
Du musst die reset_admin_pw.php aus der 0.5er Version in das Verzeichnis auch in das Verzeichnis der 0.5er Version kopieren... Die 0.3er Version wird nicht mehr benötigt

MrSmallpart, 2018-08-30:
DANKE.  Hat jetzt endlich geklappt.

Gute Nacht für heute  ;-)

MrSmallpart, 2018-08-30:
Vielleicht noch einen Frage für morgen:
Jetzt habe ich die Bauteile der alten DB in eine Neue überführt, die auf dem PC über XAMPP läuft. Ich habe aber wie schon geschrieben, auf einem externen Server die PartDB 050 auch schon eine Weile laufen, wo auch schon Bauteile erfasst sind.
Ist es möglich, die Bauteile vom XAMPP-Server mit denen vom externen Server zusammen zu führen?

jbtronics, 2018-08-31:
Das ist in der Tat schon ein größeres Problem...
Im Prinzip dürfte das mit den Import und Export Funktionen von Part-DB funktionieren...

Da die aber ziemlich unschön sind (das ist eine der Dinge die ich Mal komplett überarbeiten muss), weiß ich nicht wie gut und vollständig das aktuell schon funktioniert... Du könntest es Mal probieren (möglicherweise vorher ein Datenbankbackup machen) und dann berichten...

MrSmallpart, 2018-08-31:
Hallo Jan,

also ich habe erst einmal sicherheitshalber einmal ein neues, nacktes PartDB 050 auf dem externen Server installiert. Auch mit dem "Ventor"-Ordner. Dann habe ich versucht a.) durch Export vom "Heimserver" und Import am externen Server, da habe ich eine Fehlermeldung bekommen. Dann habe ich b.) mal versuch, die Data-Ordner zu tauschen, aber da hat er dann die Datenbank nicht mehr gefunden und ich mußte den "Original-Data-Ordner" wieder zurück kopieren. 
Sachstand ist jetzt:
eine Part-DB von 0.3.0 auf 0.5.0 updatet auf dem XAMPP-Server im PC,
eine Part-DB 0.5.0 mit einigen (vielleicht 100) Bauteilen auf externen Server und
eine Part-DB 0.5.0 neu installiert, ohne Bauteile auf dem externen Server.

Mein Wunsch wäre, die Bauteile auf dem XAMPP-Server (ca 500) mit den ca. 100 auf dem externen zusammenführen.

Gruß
MrSmallpart

jbtronics, 2018-08-31:
Was für eine Fehlermeldung genau zeigt er beim Versuch des Imports an?

Ist die Struktur mit den Kategorien, Lagerorten etc. Auf beiden Servern gleich? Weil dann könnte man evtl beide Datenbank manuell zusammenführen (wenn du bereit wärst mir SQL-Dumps der beiden Datenbanken zu schicken, kann ich Mal schauen ob ich das hinbekomme)... Möglicherweise muss man bei der Methode nochmal nacharbeiten und Kategorie und Lagerort reparieren, aber es wäre zumindest weniger Arbeit als alles nachzutragen...

Das Austausch der Data/ Ordner bringt nichts, da dort keine Infos über Bauteile drin liegen, das steht alles in der Datenbank...

MrSmallpart, 2018-08-31:
so, sind raus. Hoffe alles richtig gemacht :-)

jbtronics, 2018-08-31:
Hab die Dateien bekommen, ich werde sie mir die Tage Mal ansehen... Kann nix versprechen, aber vielleicht bekomme ich da was hin...

MrSmallpart, 2018-08-31:
Ich danke Dir jedenfalls für Deine Mühen, egal ob es klappt oder nicht. Ein Versuch ist es alle Mal wert.

Beste Grüße
MrSmallpart

MrSmallpart, 2018-09-15:
Der Vollständigkeit halber der Nachtrag:
Es hat geklappt, das hat mir sehr viel Tipparbeit erspart.
Vielen Dank.
Das eingangs erwähnte Problem wird sich vielleicht mit der Zeit, bzw. Updates auch noch bessern. ;-)

Ausdrücklicher Dank an Jan.
Beste Grüße
MrSmallpart

jbtronics, 2018-10-04:
So ich habe mal was neues probiert... Vielleicht funktioniert es ja jetzt korrekt (Änderungen sind zu finden im master Branch)

MrSmallpart, 2018-10-04:
Hallo jbtronics,
ich denke, es liegt an meinen fehlenden Kenntnissen von Java, aber das, was ich geändert habe, hat keinen Erfolg gebracht (aber auch keinen Absturz oder so).
Schon allein die Zeilennummern stimmen nicht, sie sind bei mir ziemlich am Ende von 900....
Hab ich eine falsche Version?  (Version: 0.5.0-dev).

[orig.zip](https://github.com/Part-DB/Part-DB/files/2448045/orig.zip)

MrSmallpart, 2018-10-04:
Wohl zu früh gefreut.....
Jetzt ist das Problem mit dem Kommentar nicht mehr gegeben, aber irgend etwas hab ich vielleicht doch nicht richtig geändert. Jedenfalls wenn ich nun bei einem Datensatz etwas ändern will, fehlen mir die DropDown-Felder. Ich kann es zwar über "Suchen und Hinzufügen" ändern, aber sehe die Änderung nicht. Erst wenn ich das Bauteil wieder aufrufe, dann sehe ich die Änderung.
[image]

jbtronics, 2018-10-04:
So ich habe mal deine Änderungen die relevant waren, in die aktuelle Version integriert...
Ich konnte das Problem mit den fehlenden Select feldern nicht beobachten, und eigentlich sehe ich kein Grund wieso das durch deine Änderungen auftreten sollte... (wobei ich es jetzt auch an der aktuellsten Version getestet habe, da gab es einiges an Änderungen, im Anzeigeframework...

Ich kann leider nicht wirklich beurteilen, ob das problem mit den Tooltips jetzt endgültig behoben ist, weil ich das Problem leider nur sehr sporadisch reproduziert bekomme (ich vermute das es mit den Timing zusammenhängt, wie die Funktionen aufgerufen werden, was bei jedem Aufruf schwankt, und auch von z.B. der aktuellen Auslastung zusammenhängt), daher wäre es interessant zu sehen, ob es bei anderen mit den aktuellem Patch noch auftritt..

jlegen, 2018-10-06:
> Ich kann leider nicht wirklich beurteilen, ob das problem mit den Tooltips jetzt endgültig behoben ist, weil

ist es leider nicht; bleibt bei meiner Installation bei ungefähr jedem 3. Versuch stehen. Kann ich z.B. per aktuellem Firefox auch problemlos auf http://part-db.bplaced.net "erzeugen":

[image]

jbtronics, 2018-10-06:
Ich habe eben noch ein paar Änderungen an den Tooltips vorgenommen, vielleicht klappt es ja jetzt...

jlegen, 2018-10-06:
Jo, gerade schnell getestet - diesmal sieht's gut aus! Das Tiptool springt zwar etwas "unkonventionell" nach oben links ins Eck, ist dann aber bislang zuverlässig weg!
Thx, je mehr sich meine DB gefüllt hatte, desto lästiger wurden die hängenden Dinger nämlich... :)