Part-DB/Part-DB-legacy #34: Update Anleitung

maxholgasson, 2018-10-17:
Hallo, 

gibt es irgendwo eine Update Anleitung, wenn man es per Docker Container installiert hat? habe momentan 0.5.0-dev am laufen

jbtronics, 2018-10-17:
Also wenn du die Docker-compose dateien benutzt, sollte es mit
`docker-compose up -d --no-deps --build partdb` funktionieren...

Damit dürftest du aus dem zu der aktuellsten Version aus dem master-Branch kommen (0.6-dev), was momentan identisch mit der 0.5.0 Version ist.

Du könntest auch probieren, in der docker-compose.yml die Zeile "jbtronics/part-db:latest" durch "jbtronics/part-db:stable" zu ersetzen, und dann obigen Befehl auszuführen, dann bekomst du warscheinlich die 0.5.0 Version. (Das habe ich aber nicht getestet).

Wenn das klappt ergänze ich das im Wiki.

maxholgasson, 2019-07-02:
Hallo,
ich will mich noch einmal melden, weil ich es beim ersten Versuch nocht hinbekommen habe und länger versucht hatte.
Ich habe Docker auf meinem NAS installiert und Part-DB  einfach aus der NAS Bibliothek ausgewählt. Sonst nichts. Habe auch keinen Zugriff auf eine Konsole etc. Wie könnte man das denn updaten noch?
Ich wüsste nicht wo ich irgendwelche Befehle eingeben könnte