Part-DB/Part-DB-legacy #38: Add alternative part numbers

miningandhydraulics, 2018-12-06:
Is it possible to add alternative part numbers for the same manufacturer? And make the supplier part number searchable

jbtronics, 2018-12-06:
The supplier part number is already searchable (at least if the checkbox is set under search-options).

What do you mean exactly with "alternative part numbers for the same manufacturer"? Do you have an example for that?

If you mean something like different part numbers for things like other footprints, then i am afraid, that this is not possible yet. But its one of my next big goals, to be possible, that a part can have multiple storelocations, footprints, and manufacturers, each one with its own possible part name.

jbtronics, 2022-12-18:
The symfony based Part-DB (https://github.com/Part-DB/Part-DB-symfony) has the possibility to define a manufacturer part number and a internal part number for each part.

closed 2022-12-18 by jbtronics