Part-DB/Part-DB-legacy #40: Auffälligkeiten beim Bauteile Import

christophfroehlich, 2018-12-07:
Hallo! Ich teste gerade erstmals Part-DB (0.6.0-dev, PHP 7.1.23) und prinzipiell gefällt mir das Konzept ganz gut.

Ich habe heute versucht unser bisheriges Inventar in eine aktuelle Installation von Part-DB (0.6.0-dev, PHP 7.1.23) zu importieren. Dabei sind mir folgende Sachen aufgefallen:
- Das Hochladen größerer CSV-Dateien selbst funktioniert zwar noch, beim "Daten übernehmen und prüfen" lädt die Seite sofort neu, ohne Fehlermeldung und ohne importierter Teile. Die Grenze dürfte so bei 220 Zeilen liegen, aufgeteilt auf 2 kleinere Dateien klappts mit den gleichen Daten. Ist das eine Beschränkung vom Server oder kommt das von Part-DB?
- Könnte man die Buttons für "Daten übernehmen und prüfen" und "Daten importieren!" nicht oben platzieren, sodass man nicht zweimal hintereinander ganz ans Ende scrollen muss?
- Wird beim Import irgendein automatisches Matching der Footprints gemacht? Wenn nein, dann existiert hier ein Bug. Importiere ich das File picolock.csv aus dem Anhang scheint alles zu klappen, exportiere ich das größere File inventar_part1.csv dann haben alle Picolock-Einträge plötzlich eine Random footprint ID. [inventar.zip](https://github.com/Part-DB/Part-DB/files/2658995/inventar.zip) Bei zumindest einem anderen Teil ist mir das auch aufgefallen, da bei einem Quarz plötzlich eine Diode nebenbei abgebildet war ;) Wenn ich den importierten Daten nicht vertrauen kann, disqualifiziert sich Part-DB für mich leider. Händisch anlegen ist keine Option bei so einem großen vorhandenen Inventar.

jbtronics, 2018-12-08:
Das mit dem Problemen mit den großen Dateien, könnte möglicherweise mit einer Upload Grenze des Servers zusammenhängen (In den Einstellungen gibt es einen Eintrag "Maximale Dateigröße beim Upload:", wo man die Grenze anzeigen lassen kann).

Ich befürchte aber eher, dass es irgendwo im Importsystem ein Timeout oder Fehler gibt, und es so zu dem beschriebenem Verhalten kommt. Das Problem ist, dass das Importsystem vermutlich noch aus den aller ersten Part-DB stammt (es ist noch komplett prozedural geschrieben, also stammt es vermutlich aus PHP4 Zeiten), und eigentlich ziemlich broken ist. Die von dir beschriebenen Probleme werden sich also vermutlich nicht so leicht fixen lassen.

Es steht schon seit längerem auf meiner TODO-Liste das Import (und Export) System komplett neuzuschreiben, aber dafür muss ich mindestens noch die internen Umbauten abschließen, an denen ich momentan arbeite.

Das mit den Buttons oberhalb der Daten ist eine ganz gut Idee, das kann ich auch noch leicht umsetzen.

christophfroehlich, 2018-12-08:
Die "Maximale Dateigröße beim Upload" sind 256MB, an dem wirds nicht liegen. Die Daten werden auch verarbeitet, weil die Liste ist ausgefüllt bevor ich "Daten übernehmen und prüfen" klicke. Nur danach sinds halt eben weg.

Das mit den Buttons ist nur eine Kleinigkeit. Hab nur beim ersten Mal die Buttons nicht gefunden und wusste nicht, was jetzt passiert ;) Wenn mans dann weiß, ists eh kein wirkliches Problem.

Das mit dem Importsystem hab ich schon bei dem andren Issue zum Baugruppenimport gelesen. Ich warte gespannt auf ein Update :)
Mal schauen, ob ich nicht lieber in der Zwischenzeit die Sachen händisch per phpmyadmin importiere.

jbtronics, 2018-12-08:
> Die "Maximale Dateigröße beim Upload" sind 256MB, an dem wirds nicht liegen. Die Daten werden auch verarbeitet, weil die Liste ist ausgefüllt bevor ich "Daten übernehmen und prüfen" klicke. Nur danach sinds halt eben weg.

Wenn das so ist, liegt das vermutlich an der Variablen beschränkung von PHP (dafür gibt es die Anzeige "Max. Input Vars:" in den Einstellungen). In der PHP.ini gibt es eine Einstellung "max_input_vars", wenn man die hochsetzt wird es vermutlich auch mit größeren Eingabedateien klappen.

Das Problem ist, dass jedes einzelne Feld in der Tabelle, eine einzelne Variable ist, die an den Server übertragen wird, und PHP eben diese Beschränkung hat. Da werde ich mal schauen ob ich eine bessere Methode für finde, wenn ich das import system neu schreibe...

christophfroehlich, 2018-12-09:
Max. Input Vars sind bei mir auf 3000 gesetzt, das könnte sich dann in etwa ausgehen.