Part-DB/Part-DB-legacy #42: Zuweisen von Footprints für mehrere Artikel

christophfroehlich, 2018-12-12:
Hallo!

Ich bin nach wie vor am Importieren des vorhandenen Inventars in Part-DB 0.6.0-dev.

Das Zuweisen von Kategorien oder Hersteller aus einer Bauteilliste (eine oder mehrere Zeilen markieren, "Verschieben nach", Kategorie/Hersteller auswählen, Ok) funktioniert wie gedacht. Das Zuweisen von Footprints oder Lagerorten auf die gleiche Art und Weise jedoch nicht, ohne jeglicher Fehlermeldung.

Hat das einen bestimmten Grund?

jbtronics, 2018-12-14:
Gibt es in der Konsole deines Browsers irgendwelche Meldungen oder Fehlermeldungen?

Geht das bei dir in der Demo (http://part-db.bplaced.net) korrekt? Weil in meiner Installation und in der Demo funktioniert das problemlos.

christophfroehlich, 2018-12-14:
Habe es jetzt nochmal probiert, tatsächlich: in der Bauteilliste  `show_category_parts` funktionert das mit Kategorie/Hersteller/Footprints/Lagerort gleichermaßen wie gedacht.

Nicht klappen tut es zB, wenn ich mir alle Bauteile eines Lagerorts anzeigen lasse mit `show_location_parts` oder via `show_footprint_parts` und dann Kategorie/Hersteller/Footprints/Lagerort ändern will. 

Also anders als oben beschrieben. Sorry, da war ich dann wohl verwirrt mit den verschiedenen Listen. Gleiches verhalten bei 0.6.0 und 0.5.5 in der Demo.

christophfroehlich, 2018-12-14:
Nachtrag, Fehlermeldungen in der Console gibts:
[image]

christophfroehlich, 2018-12-14:
kann ich bestätigen, danke 👍