Part-DB/Part-DB-legacy #44: System -> Konfiguration: Einstellungen werden nicht übernommen

christophfroehlich, 2018-12-13:
Die Änderungen in der Systemkonfiguration (zb des Themes, oder die Häkchen bei Aussehen, Funktionen, Sonstiges, Entwickler) werden beim Klicken auf "Einstellungen Übernehmen" wechselweise: 

- gar nicht übernommen oder

- übernommen, aber der Status der Häkchen ist im ersten Moment falsch. Erst wenn die Seite händisch neu geladen wird stimmen auch die Häkchen.

Einmal hatte ichs auch zusammengebracht, dass der Vorgang (Einstellungen setzen, dann den Tab wechseln, dann erst auf "Einstellungen übernehmen") die Einstellungen übernommen hat, ohne den Tab zuwechseln reproduzierbar jedoch nicht.

Das ganze wurde mit dem admin-user sowie eines anderen Users mit admin-Rechten auf 0.6.0dev beobachtet.

jbtronics, 2018-12-14:
Was für einen Browser benutzt du denn? Hast du JavaScript aktiviert?

Weil normalerweise sollte der Browser die Website automatisch aktualisieren, wenn auf "Einstellungen übernehemen" geklickt wurde. Und dann sollten die von dir beschriebenen Effekte nicht auftreten.

Freshdouble, 2018-12-14:
Bei mir treten die selben Probleme auf, Betriebssystem Windows 7, Browser Firefox 64.0.

Die Webseite wird beim klick auf "Einstellungen übernehmen" neu geladen, aber anscheinend werden die Einstellungen nicht übernommen.

Javascript ist aktiviert. In der Browserkonsole taucht folgende Fehlermeldung auf:

[Exception... "Favicon at "https://*****/icons/apple-touch-icon.png" failed to load: Not Found."  nsresult: "0x80004005 (NS_ERROR_FAILURE)"  location: "JS frame :: resource:///modules/FaviconLoader.jsm :: onStopRequest :: line 156"  data: no]

Wo werden die Einstellungen gespeichert? Dann prüfe ich ob nur die Änderung im Webinterface nicht sichtbar wird oder ob die Einstellungen gar nicht geändert werden.

jbtronics, 2018-12-14:
@Freshdouble die Einstellungen werden in data/config.php gespeichert...

Die Fehlermeldung die du gepostet hast, sollte aber nichts damit zu tun haben...

Könnt ihr mal probieren den Inhalt des Ordners templates_c zu löschen (das ist der Cache Ordner für die HTML ausgaben), vielleicht hängt das damit irgendwie zusammen...

Freshdouble, 2018-12-14:
In welchem File ist denn die Smarty Config?
Wo wird die Seitenausgabe erzeugt?

Ich werde mal nachsehen worans liegt