Part-DB/Part-DB-legacy #48: Featureanfrage: zeige alle Bauteile ohne Lagerort

jlegen, 2018-12-14:
Täte zumindest bei meinem Workflow helfen: erst alles erfassen, später/danach vom „Haufen“ in die diversen Magazine...

christophfroehlich, 2018-12-17:
Funktioniert der Workflow nicht jetzt eigentlich schon wenn man halt einen Lagerort Import oder "Haufen" erstellt?

Ähnlich dazu hätte ich auch eine Featureanfrage: Anzeige aller Lagerorte ohne zugeteiltem Bauteil. Szenario: Ich habe 300 einzelne Boxen in einem Kleinteilesortiment, diese alle als Lagerorte hinzugefügt (zB. A0 - AD9) und möchte nun die leeren Fächer mit Bauteile befüllen.

jlegen, 2018-12-18:
> 
> 
> Funktioniert der Workflow nicht jetzt eigentlich schon wenn man halt einen Lagerort Import oder "Haufen" erstellt?

Hast recht, das wäre natürlich ein sehr einfacher Workaround. 👍 

> 
> Ähnlich dazu hätte ich auch eine Featureanfrage: Anzeige aller Lagerorte ohne zugeteiltem Bauteil. Szenario: Ich habe 300 einzelne Boxen in einem Kleinteilesortiment, diese alle als Lagerorte hinzugefügt (zB. A0 - AD9) und möchte nun die leeren Fächer mit Bauteile befüllen.

...und pro Box dann ein Bauteil? Denn nach dem ersten zugeordneten würde diese ja aus der Liste fliegen... :)

christophfroehlich, 2018-12-18:
> ...und pro Box dann ein Bauteil? Denn nach dem ersten zugeordneten würde diese ja aus der Liste fliegen... :)

Doch, so ists gemeint. In unserem Sortiment liegen pro Fach nur idente Bauteile. Somit ist der Lagerort nicht mehr leer, nachdem auch nur 1 Exemplar einsortiert wurde. Es ist aber auch nicht voll, weil noch 500 weitere Exemplare Platz hätten ;)

jbtronics, 2023-07-19:
Possible with the filter system in the Part-DB rewrite: https://github.com/Part-DB/Part-DB-server

closed 2023-07-19 by jbtronics