Part-DB/Part-DB-legacy #49: Lieferanten Webseite hinzufügen akzeptiert URL nicht

pboe, 2019-01-17:
Beim Hinzufügen oder Ändern von Lieferanten werden URLs wie z.B. www.foo.bar mit dem Hinweis please enter URL abgelehnt. Feld ist rot markiert. Ebenso bei Artikel Direkt Link.

"please enter URL" ist auch noch nicht übersetzt bei aktivierten deutschen Spracheinstellungen, allerdings nur im Firefox, nicht im Safari.

Part-DB 0.5.6 (stable)
Firefox 66.0a1 (2019-01-12) MacOS 10.13.6
Safari Version 12.0.2

jbtronics, 2019-01-17:
Das mit der fehlenden Übersetzung ist ein Problem vom Firefox (die Warnung, dass es keine URL ist, wird vom Browser selbst erzeugt, da habe ich keinen Einfluss..)

In dem Kontext verwende ich das HTML URL Feld, und scheinbar muss da ein http:// bzw. https:// vorkommen. Als workaround musst du also nur https:// vor deine URL setzen.

Ich werde aber Mal probieren, ob ich dass ändern kann, so dass das nicht mehr unbedingt nötig ist.

jlegen, 2019-01-17:
Naja, "works as designed", würde ich sagen - zu einer **URL** gehört das Protokoll ja per Definition. Sonst könnte man das auch einfach als "Textfeld" anlegen...