Part-DB/Part-DB-legacy #53: Enhancement: Bauteile/Baugruppen abfassen mit Kommentar versehen

pboe, 2019-01-24:
Hi,
die Möglichkeit Baugruppen zu definieren ist toll.
Nützlich wäre, wenn man beim Abfassen der Teile einer Baugruppe, eigentlich auch bei der Entnahme einzelner Bauteile, einen Kommentar hinzufügen könnte, so dass  man verfolgen kann wohin die Bauteile geflossen sind.

pboe, 2019-01-24:
Danke für die schnelle Antwort.
Ah, das liegt an dem Theme superhero.min.css
Unter Firefox 66 sowie Safari 12.0.3 auf MacOS werden beim Entnehmen eines Bauteils sowohl der Entnehmen Button wie auch der Pfeil nur angezeigt, wenn man mit der Maus drüber hovert. Mit dem Default Theme ist es direkt sichtbar.

Bei der Baugruppenentnahme wird die Baugruppe  automatisch als Kommentar eingefügt. Wünschenswert wäre aber ebenso ein manueller Kommentar. Z.B. Seriennummer, Kunde ...

pboe, 2019-01-24:
[image]

jbtronics, 2019-01-24:
Da passt die button Farbe wohl nicht zum Hintergrund...
Ich teste eigentlich immer nur mit dem Default theme, daher ist mir das nicht aufgefallen.

Ich schaue Mal ob ich eine Lösung finde, dass es auch mit diesem theme sichtbar ist

pboe, 2019-01-24:
Nicht wichtig, bin jetzt auch wieder auf dem Standard Theme.
Wenn Du eh mal den Code mit den Baugruppen anfasst, würde ich mich freuen, wenn man da noch nen eigen Kommentar beim abfassen anhängen könnte.