Part-DB/Part-DB-legacy #54: Label functionality missing

MaxHausmann, 2019-01-30:
Hello again,

where can I activate the label functionality for printing barcodes etc.?
Can't find any setting to enable it.

Best regards
Max

jbtronics, 2019-01-30:
If a user can create labels or not can be controlled via permissions (under system-> users or system->groups).
You can find it under Miscellaneous and Labels.

The checkboxes has to be set for a user, that should be able to create labels.

MaxHausmann, 2019-01-30:
I checked the admins/user groups and all groups have a "-" at "Labels", which means obviously that it inherits from a group above. But there is no group to inherit from. However I tried to set the checkbox to allow this setting but it won't be saved and still remains at "-".

MaxHausmann, 2019-01-30:
Update: "special" permission for each user works, but not for a specific group.

jbtronics, 2019-03-10:
Are you sure?
In my installation and the online demo the user is able to generate barcodes, when he is in the "user" group, and the "user" group has the barcode permission (The user has its permission set to inherit).

With what version have you tested it?

jbtronics, 2019-03-10:
Mal etwas leicht anderes:

Ich habe etwas spät geschaltet, aber falls ihr Part-DB noch nutzt und Hilfe brauchen solltet, ich bin auch zufälligerweise Student in Jena...
Ich habe auch einen starke Vermutung, in welchem Makerspace ihr das laufen habt xD

MaxHausmann, 2019-03-10:
We are currently using version 0.5.6 (stable). I installed accidentally first some nightly build and than the stable one, maybe there are some data fragments of the old installation.

german:
Ja wir sind gerade dabei, Part-DB komplett mit Daten zu befüllen. Hab mit so vielen Leuten zu tun, aber Dein Name kam mir schon bekannt vor. Ja Lichtwerkstatt am ACP. Du bist der, der sich um die Lackschleuder kümmern wird? Wir können sonst auch mal privat weiter schreiben oder wir quatschen mal persönlich.

jbtronics, 2019-03-10:
Ja der bin ich (ich hoffe dass ich da nächstes Semester endlich Zeit für habe)...

Du kannst mir gerne Mal eine E-Mail schreiben (sollte in meinem Profil stehen), dann müssen wir hier nicht die Issues vollschreiben. Ich bin aber vermutlich erst wieder in ein paar Wochen wieder in Jena, dann kann ich aber gerne Mal vorbeikommen

alarmschaben, 2019-03-11:
Moin!

Gestern bin ich aus akuter Ordnungswut über dieses tolle Projekt gestolpert und habe mich dafür entschieden.

Ich würde gern in den nächsten Tagen einen PR für Datamatrix- (und wohl auch QR-) Support einreichen. Oder arbeitet von Euch gerade aktuell jemand daran?

Viele Grüße
Armin

jbtronics, 2019-03-11:
@alarmschaben Hallo Armin,
soweit ich weiß arbeit da niemand im Moment dran. 
Ich arbeite momentan daran, eine Art "Rewrite" von Part-DB auf Basis von Symfony4 zu erstellen (https://github.com/Part-DB/Part-DB-symfony), da die aktuelle Codebasis leider sehr unsauber ist, und Erweiterungen immer nur mit großen Aufwand möglich sind. Außerdem ist es durch die Symfony4 Basis dann einfach möglich, Dinge umzusetzen, die mit der aktuellen Codebasis quasi unmöglich sind (z.B. kann man jetzt einfach zwischen den Sprachen wechseln, indem man /de/ bzw. /en/ vor die Routen setzt). Mein Ziel ist es, dass diese Symfony Version, irgendwann die Version dieses Repositories ersetzen kann. Momentan bin ich aber noch bei den absoluten Basicfunktionen, daher werden da noch mindestens einige Monate vergehen, bevor der aktuelle Code obsolet wird.

Wenn du also Lust hast, Support für Datamatrix und QR-Codes zu implementieren, würde ich mich sehr freuen.

MaxHausmann, 2019-03-11:
Wenn ihr noch jemand für Front-End und Design braucht, ich hätte ab Sommer/Herbst auch wieder Zeit.

alarmschaben, 2019-03-11:
Moin Ihr Zwei!

Prima, dann hat das in der Zwischenzeit (zumindest für mich) auch einen Nutzen. Wobei das Feature ja auch ziemlich einfach einzubauen ist. Die DMTX-Ausgabe tut hier bei mir schon, ich muss es nur noch schön machen, damit der Code hinterher nicht noch dreckiger wird. ;-)