Part-DB/Part-DB-legacy #55: Datumsortierung bei Zuletzt bearbeiteten Bauteilen "defekt"

darthm0e, 2019-02-11:
Die Liste bei Zuletzt bearbeiteten Bauteilen / Zuletzt hinzugefügten Bauteilen sortiert das Datum nur nach dem ersten Feld. Also nur nach den Tageszahlen. Monat bzw Jahr werden nich berücksichtigt.

Bsp:

Teil 1                       15.01.2019
Teil 2                       14.01.2018
Teil 3                       12.**03**.2019
Teil 4                       11.**02**.2019

pboe, 2019-02-26:
Kann ich mit 0.5.6 (stable) bestätigen, allerdings stimmt die Sortierung, wenn man Verwaltung -> Zeige ->  Zuletzt hinzugefügte Bauteile aufruft.
Ändert man die Sortierung über den Tabellenheader, kommt die Sortierung, wie oben beschrieben durcheinander.

jbtronics, 2019-02-26:
Das liegt vermutlich daran, dass beim Initialen Laden die Sortierung durch den SQL-Server geschieht, beim Sortieren über die Tabellenheader, das clientseitig geschieht...

Ich schau Mal ob ich das repariert bekomme...