Part-DB/Part-DB-legacy #57: Bauteil Eigenschaften erweitern

miathedev, 2019-03-16:
Hallo zusammen,

gibt es die Möglichkeit Bauteilen zum Beispiel Kategorie->Transistoren zusätzliche Spalten hinzuzufügen in Form von "gate threshold voltage" etc?

Natürlich kann ich das ganze auch theoretisch in die Beschreibung schreiben, aber eine Sortierung anhand der gate threshold voltage wäre ziemlich praktisch!

Mit freundlichen Grüßen,
Mia

jbtronics, 2019-03-16:
So eine parametrische Sortierung ist sehr cool. Das Problem ist, dass die notwendige Datenbankstruktur dafür relativ aufwendig ist, wenn man rein relationelle Datenbanken (wie z.B MySQL/MariaDB, PostgreSQL oder SQLite) verwendet.

Relativ einfach geht das mit dem JSON Datentyp (so macht das z.b. ElaLa), das setzt aber einen relativ modernen MySQL Server (5.7 oder höher) vorraus.

Momentan arbeite ich an einem Rewrite von Part-DB auf Basis von Symfony 4. Wenn das fertig ist, dürfte das deutlich besser erweiterbar sein, als die bisherige Version. Dann kann man dann auch eine parametrische Suche einbauen.

miathedev, 2021-03-21:
Hab ich mal geschlossen da jetzt am neuen gearbeitet wird!

closed 2021-03-21 by miathedev