Part-DB/Part-DB-legacy #60: login happens with wrong user

christophfroehlich, 2019-05-11:
Frequently it happens that after the successful login of user B he is logged in as user A, which was already logged in before. If this happens, there is no entry in the event log of user B after the login.

I've got no clue if this is related to server-side settings or partdb-related?

0.6.0-dev, partdb recognizes php 7.1.25, .htaccess is not configured.

jbtronics, 2019-05-13:
This could be possible related to Part-DB. The id of the user that is currently is logged in is saved in the $SESSION var of PHP. This value should be deleted, if the user logs out. But if there is a problem doing that, this value could be used again in during the login process.

I will investigate that in the code.

Is there any way to provoke that behavior or is it completly random?

christophfroehlich, 2019-05-13:
We encountered that several times if two users wants to work with part-db:
- user A logs in
- user A is working somewhere in part-db
- user B logs in on a different PC while user A is still logged in and working
- user B is logged in as user A in the interface instead of user B

jbtronics, 2019-05-25:
I have tried to reproduce that issue, and it was not possible for me (neither in 0.5.6 nor my 0.6-dev installation).
Are you sure that you have installed the current 0.6-dev Version from master? Maybe i have fixed that issue sometime ago, and you have an older version without the fix?

Are you using some modified php.ini file? Especially with changes on the session.* options? Maybe this is a php config related issue?