Part-DB/Part-DB-legacy #61: Using Docker-latest: image upload doesn't work

julianrendell, 2019-06-07:
Thank you for creating part-db! 

We are trying it for inventory management of tools and components (not just electronics) in our makerspace.

However image upload for parts appears to be broken in the "latest" docker image.

Followed the docker instructions for the "latest" option.

Created a basic category and location. 

Created a part.

Edited info for part, tried to upload an image from phone, and from PC. Failed ("Name can't be empty", or nothing happens.)

Tried on Safari and Chrome.

Re-deployed using stable image, and everything is working beautifully (thanks!)

Side question: any recommendation re entering equipment (soldering irons, ovens, etc)?

OpaApo, 2019-06-21:
I would also like to thank you for creating Part-DB.

We didn't use Docker but installed it the normal way. It's 0.6.0-dev from 04.05.2019.
The "Browse" button works. After choosing a file, the "Remove" button appears but the field "Dateiname / URL:" stays empty. Nothing happens when pressing "Hinzufügen". Tried on Firefox and Chromium.

jlegen, 2019-06-25:
had the same issue; fix for me was:
in file inc/lib.php, swap this if-clause (~ line 283)

    //Dont allow to upload a PHP file.
    if (strpos($file_array['name'], '.php') != false
        || strpos($destination_filename, '.php') != false) {
        throw new \Exception(_('Es ist nicht erlaubt PHP Dateien hochzuladen!'));
    }

with this one below:

    if ($destination_filename == null) {
        $destination_filename = $file_array['name'];
    }

(=> make sure $destination_filename is set before applying strpos() on it)

OpaApo, 2019-06-27:
> had the same issue; fix for me was:
> in file inc/lib.php, swap this if-clause (~ line 283)
>
> ```
> //Dont allow to upload a PHP file.
> if (strpos($file_array['name'], '.php') != false
>     || [...]

We incorporated it and it works. Thank you!

jbtronics, 2019-07-12:
I switched the if clauses in the file, so it should work now for everyone.

closed 2019-07-12 by jbtronics