Part-DB/Part-DB-legacy #63: Backups / Restore

NetForces, 2019-06-13:
Amazing project !

How can we backup when running in a docker instance ?

Is everything in the MySQL database ? Images also ?

jbtronics, 2019-06-13:
The data about the parts is saved in the MySQL DB.
Images and Attachments (and the main config) is saved under data/ (this folder is persistent in docker).

NetForces, 2019-06-13:
Perfect, I can put a backup script together.

Might be a nice feature to add. Maybe a configurable periodic backup of the database and images to an external target. Maybe using scp or something similar.

mrWheel, 2021-05-03:
> Perfect, I can put a backup script together.
> 
> Might be a nice feature to add. Maybe a configurable periodic backup of the database and images to an external target. Maybe using scp or something similar.

Hi I’m new with docker so have no idea where to start. Have you made a script for it and are you willing to share?

thanks in advance!