Part-DB/Part-DB-legacy #64: Scan labels to change location

rs-development, 2019-07-11:
I would love to have a feature of scanning labels to change storage location.

Do you think about implementing a API or similar, i could do the client side implementation.

jbtronics, 2019-07-12:
Implementing an API for the Part-DB version in this repo is difficult (you would have to implement every endpoint manually, and the auth system can not really handle auth tokens).

However I am working on a completly new Part-DB version based on symfony4 (see https://github.com/Part-DB/Part-DB-symfony for repo). Because it uses symfony and doctrine, it should be very easy to implement API Platform (https://api-platform.com/), which allows to expose the models of the database via powerful standardized APIs (like REST or GraphQL).

If you like i can try to implement a simple API using API Platform that could be used for external applications. But please mind that the symfony Part-DB version is far from being complete (the internal models used for the API should work though), and i dont know how stable it is (you can use it in parallel to the old Part-DB version on the same Database)

rs-development, 2019-07-12:
Okay, then i am maybe trying around with client itself a bit and i will make a simple docker instance which takes the API requests from the client and directly modifies the database.

If it works and proves to be useful we can take a lot look how the API developed and how to integrate it into the symphony version

rs-development, 2019-07-19:
So i have implemented a go server application listening for http requests and i am on the android app right now. While i tried to create labels i recognised that i cannot create labels with the dev branch. There is no such option.

What are you running on the demo site?

rs-development, 2019-07-19:
And i also recognised, or lets say the app, that the storage location barcodes are of type CODE39 and not EAN8 like the parts. Maybe change that in the barcode selection field.