Part-DB/Part-DB-legacy #70: After update to 0.5.7: SQLSTATE[HY000]: General error: 1525 Incorrect TIMESTAMP value: '-1'

jsuelwald, 2019-10-29:
Hi,

i've updated an existing installation of PartDB to 0.5.7 and on /system_log.php this error occured:

Datenbankfehler:
SQLSTATE[HY000]: General error: 1525 Incorrect TIMESTAMP value: '-1'

SQL-Query:
SELECT COUNT(id) AS count from log WHERE level <= ? AND (datetime >= ?) ORDER BY log.datetime DESC

Parameter:
Array
(
[0] => 7
[1] => -1
)

jbtronics, 2019-10-31:
What Part-DB version has you used before? What MySQL/MariaDB server version are you using?

corn11, 2022-12-01:
Hi,
I've experienced the same issue with PartDB 0.5.12 and MySQL 8.0.31.
The problem seems to be that MySQL 8 doesn't accept the string value '-1' as a valid timestamp only if it isn't a string (I've verified this with an older MySQL version)

I've solved the (possible) issue(s) by editing the "filter for dates" of the getEntries() and getEntriesCount() functions in the lib/Log.php:

//Filter for dates
if (($max_date != "") **&& ($max_date != "-1")**) {
    $query .= " AND (datetime <= ?)";
    $data[] = $max_date;
}

if(($min_date != "") **&& ($min_date != "-1")**)
{
    $query .= " AND (datetime >= ?)";
    $data[] = $min_date;
}`

jbtronics, 2022-12-18:
Should be fixed in Part-DB 0.5.13

closed 2022-12-18 by jbtronics