Part-DB/Part-DB-legacy #72: Undefined index: perms_labels

chrisnoisel, 2019-12-20:
Hello,
I've have a setup of the current master branch on apache/mariadb, database is version 26.
I noticed in my log that php complains about a missing "labels" permission. I looked it up, here what happen :

the admin user, coming from a fresh partdb install, has a "perms_labels" set to 0, so every request to label permission is translated to being group-inherited ($val === BasePermission::INHERIT)
Unfortunately the database table "groups" doesn't include a "perms_labels" column, thus the warning is generated.

This could be addressed by creating a new database version, but looking at update scripts, I'm unsure how to properly generate a migration. This doesn't look too critical (basically, labels permissions are disabled), is this worth fixing right now ? and if so, can you help ?

jbtronics, 2020-01-05:
I am not sure if I will fix that. When I change the DB structure I have to change the Migrations of Part-DB 1.0 too (they are currently targeting the structure of Part-DB 0.5.6).
The label permissions are fixed in Part-DB 1.0 and they are not really needed (at least they are not security relevant)...

Any other opinions on this?

chrisnoisel, 2020-01-05:
Looking at your commit activity I figured you would rather focus one the 1.0, which is totally fine.
If partdb 0.5.6 is dev-deprecated then it would be useful to have like a roadmap on the current 1.0 development (what is done, work-in-progress, etc.) or some insights about what you're going to do. I may have some features I'd like to add but right now, I've found it's harder to dig in the new version (well, not knowing Symfony doesn't help, so I've some homework to do on my own)

jbtronics, 2020-01-06:
After you have understood some basic ideas of Symfony the Part-DB 1.0 codebase should be easier to understand than the version in this repo (also the development is so much more comfortable).

I have written something about my plans for Part-DB and my ideas for a roadmap in the new repo (thanks for your suggestion): https://github.com/Part-DB/Part-DB-symfony/issues/9