Part-DB/Part-DB-legacy #74: Purchase Order

a7hybnj2, 2020-01-01:
I really like the look of the DEMO site and am thinking of using this for my inventory needs. My question is:

How to best enter parts from a purchase order?
 - Would that be by tools - import - csv? Which means I need to make a csv file from the PO or do I need to go to each part and add stock?

a7hybnj2, 2020-01-01:
I think I might have found it. It doesn't look like you can make/add a PO but you can just see a list of parts that need to be order per supplier. Then I guess you just have to check the ADD STOCK box when they show up.

Would be nice if there was a check box for 'ordered' or 'awaiting arrival' or 'on PO' or something. So, if you are waiting for ordered parts to arrive but your getting ready to place a 2nd order there isn't just one big list of everything on there.

jbtronics, 2020-01-01:
The whole ordering and project management in the old Part-DB version is really bad. That is some of the thing I will remake completely in the Part-DB 1.0 version. Therefore I will not try to fix it in this repo.

You should open this as feature requests in the new repo, so that I can implement the features there.

closed 2020-01-01 by jbtronics