Part-DB/Part-DB-legacy #76: How to add Euro (€) symbol as currency in english language

glaukos78, 2020-01-19:
How to add Euro (€) symbol as currency in english language (en_US)

jbtronics, 2020-01-20:
In the Part-DB version in this repo, the currency symbol of the configured locale is used, so en_US will not work.

You could try to replace the value for "$config['language']" in the data/config.php file with a english locale with an european country like en_DE or en_GR. You maybe have to add these locales to your system before.

In the Part-DB version I am currently working on (https://github.com/Part-DB/Part-DB-symfony), it will be possible to set any currency you like, independently from the locale. Furthermore you can define other currencies, so that you can save price informations in different currencies with automatic conversion to your local currency.

glaukos78, 2020-01-20:
Thanks for answer.
I have change value for "$config['language']" in the data/config.php with **en_IE** and working fine.

closed 2020-01-20 by glaukos78