Part-DB/Part-DB-legacy #77: Import CSV issue

spi5498, 2020-01-21:
After uploading the file and checking that the data is OK, clicking "Apply data and check" results in it just returning to "Import", and nothing gets imported. Also no error message.

jbtronics, 2020-01-21:
How do your CSV file look like?

What happens if you try to upload the example files: https://github.com/Part-DB/Part-DB/tree/master/documentation/examples/import_parts

spi5498, 2020-01-21:
Actually, importing the example, results in a green message with a green checkmark, and the text "Invalid data". So that is pretty strange?

jbtronics, 2020-01-21:
What Part-DB version do you use?

spi5498, 2020-01-21:
0.5.8

jbtronics, 2020-01-21:
Okay, I have just released a new version, where import should work as intended: https://github.com/Part-DB/Part-DB/releases/tag/v0.5.9

spi5498, 2020-01-21:
Just tested it right now, and the example file works. A blue button appears after pressing "apply data and check", however the problem with my file is still the same.
After pressing "Apply data and check" it disappears and returns to the "import" without any messages.

Can i send you my file in some way? The strange thing is that it shows up OK after "upload"

jbtronics, 2020-01-21:
You can send me your file via mail: jan.h.boehmer@gmx.de

And i will look into it if I have time

jbtronics, 2020-01-21:
Okay, I found the problem.
The issue is related to a limit of PHP, that only some amount of variables can be passed via request.
Every field you see in the table will be transfered as form field, so the default limit is reached early.

To change that: Go to your php.ini file, search for `max_input_vars` and change it to `max_input_vars = 3000`. Then restart your webserver and the import should work.

In the new version I am working on, I hope to find a better solution, so that even big imports will not cause problems...

spi5498, 2020-01-21:
Did you try that solution on your end? Because i must have done something wrong. It is still the same.
Also tried to just cut the file in two parts instead. Same thing

jbtronics, 2020-01-21:
What is shown in System->Config page in the server panel for the value of "Max. input vars" ?
When the value there is less than 3000 your change to ini settings was not applied.

Splitting the file in half is not enough (it is still too long then), maybe try to use just the first 5 rows or so...

spi5498, 2020-01-21:
Youre absolutely right. It just says 1000.

I did check that /etc/php/7.3/cli/php.ini is the loaded config file. Did the edit and restarted apache. So there is something not playing along

jbtronics, 2020-01-21:
The cli folder is wrong (this one is used when you run PHP from command line). You need to edit the file under /etc/php/7.3/apache or /etc/php7.3/apache (or similar) to edit the one used by apache...

spi5498, 2020-01-21:
Yes! That did it! Thank you very much

closed 2020-01-21 by spi5498

mrWheel, 2021-05-03:
Hi,
Having the same issue with the docker-stable image, Version: 0.5.8 (stable), Git: stable-v0.5/
"NextGen".
But there is no file "php.ini" ....
Is there a solution for the docker image?