Part-DB/Part-DB-legacy #79: Bug: Type Error on spuuplier link on part page

rs-development, 2020-04-17:
I am running PartDB from your provided docker container. See below. I get the following error after clicking on Supplier Link on the procut page.

Where should this link leed?
Whats the intention of the SELECT?

I take i closer look at it later today, let me know if you need more information!

[image]
Corresponding php code:

[image]

Database entries:

[image]

[image]

Actual Container used:

{
        "Id": "sha256:f5c33f6201e0145c1c802b69496fc3d16f3d8c46cb56ee3cac52b7e80df200cd",
        "RepoTags": [
            "jbtronics/part-db:latest"
        ],
        "RepoDigests": [
            "jbtronics/part-db@sha256:7a07b1904b180040004365e8a45d5e2ee099a66c028683ed732a37338ebfc497"
        ],

rs-development, 2020-04-17:
[image]

So the query itself seems to be working IF the supplier ID is given correctly.

rs-development, 2020-04-17:
A simple cast to int fixed the error. But why the cast is needed in my case?

jbtronics, 2020-04-17:
As it is not possible for PDO know which type was returned by DB, it returns the value as string.
But as the getPartsCount() function must return an int, it fails, when trying to return some other type (like string).

This is an leftover from the 0.5.* versions (there was no return typehint, so the DB value was converted to int implicitly somewhere else). All other Elements use another base method, which already included the int cast: https://github.com/Part-DB/Part-DB/blob/0c9fa13fa697385a2db6cfbc4e814f8a03dcd0de/lib/Base/PartsContainingDBElement.php#L287

This kind of problems hopefully does not happen so often in the new Part-DB 1.0 versions, as it is developed with strict typing in mind and Doctrine does most of low level database things...

closed 2020-04-17 by jbtronics

rs-development, 2020-04-18:
Thanks for the fast response! I was in doubt whether I messed up the database or not.