Part-DB/Part-DB-legacy #80: add landing page text

manuel-91, 2020-05-08:
Is there a way to add/modify the text that is printed when I go to the landing page?

At the moment the [landing page of the demo](http://part-db.bplaced.net/part-en/startup.php) shows:
- name of the instance, current version, user login
- licence informations

It would be great if (maybe in settings) an additional text here could be edited. The text could explain things like:
- who is the maintainer of the service technically
- who maintains the db content?
- when location of a part tells "box42" or "locker404", where are they in a few more words. Whom to ask? Maybe a link to an internal wiki.
- how to buy more parts, is it done monthly, who is doing that

jbtronics, 2020-05-08:
In settings there is the possibility for some BBCode that is shown on the start page.
You can use that add custom text and links to your start page.

manuel-91, 2020-05-09:
Thank you very much!

closed 2020-05-09 by manuel-91