Part-DB/Part-DB-legacy #84: Possibilty to change currency independent of language

Vorathel, 2020-08-18:
Dear developers,
could it be possible, to change the currency without changing the language? We are working with international people and would like to keep Euro as currency and English as language. Maybe a dropdown menu could be possible for different currencies in the edit_part_info window.
Thanks a lot in advance,

Vorathel

jbtronics, 2020-08-18:
In the new Part-DB version (https://github.com/Part-DB/Part-DB-symfony) this is easily possible, you can even create multiple currency and use them simultaneously in price details.

The current Part-DB version uses the configured locale for formatting monetary values. As a workaround you can use a english locale from a country that use euro. You can try to add locales like "en_DE" or "en_BE" to this config file https://github.com/Part-DB/Part-DB/blob/e6a3529c96f0cb767a4e4ce7d8f67dfbb6274795/inc/config_defaults.php#L249. The main problem there should be that your system must have defined this locales, so you maybe have to generate them with `locale-gen`

jbtronics, 2023-07-19:
Possible in the Part-DB rewrite: https://github.com/Part-DB/Part-DB-server

closed 2023-07-19 by jbtronics