Part-DB/Part-DB-legacy #89: Labels and Profiles Question

EvansMike, 2020-12-30:
I am wanting to keep my custom barcode label settings and I assumed I could do this via the profiles tab.  Nothing happens when I save or load a profile.  Have I misunderstood something or have I mis-configured something.  I have ensured I have the correct permissions in Sytem->user settings.
Where are the profiles stored?  Maybe I can manually edit them.

EvansMike, 2020-12-30:
It looks like the initial data/label_profiles.json isn't being created.  
Plus a error in editItem(...) params which prevent changes being saved.

Peter-Krebs, 2021-06-11:
This is certainly a bug. The barcode label settings can not be saved.

The status of the save request is HTTP 200 OK, but prints out an error message anyway:  
> Argument 2 passed to PartDB\Tools\JSONStorage::editItem() must be of the type bool, array given, called in /var/www/html/show_part_label.php on line 263
> (Exception was thrown in /var/www/html/lib/Tools/JSONStorage.php, Line 178)

The function expects `$data` to be bool instead of any object, this is certainly a bug:
```php
public function editItem(string $key, bool $data, bool $write_data = true, bool $create_when_not_exist = false)
```

The expected type should be changed to `mixed` or to nothing (like `JSONStorage::addItem()` does) as JSON can encode any type (see [PHP Docs: Type declarations](https://www.php.net/manual/en/language.types.declarations.php)).

As you have not received any feedback for this error the only solution would be to fork the repository and fix it yourself.

EvansMike, 2022-01-12:
That fixed branch link is now https://github.com/EvansMike/Part-DB/tree/issue%2389

jbtronics, 2022-01-13:
Hey thank you for fixing the bug. Can you create a pull request? Then I will merge it into the master branch.

EvansMike, 2022-01-13:
PR created.

jbtronics, 2022-01-19:
I merged the pr, should be fixed in newest master branch.

closed 2022-01-19 by jbtronics