Part-DB/Part-DB-legacy #92: First install doesn't create database and error code pop up.

BjornOgren, 2021-02-19:
Hello! When i try to install Part-DB i will say that install is complite, but it do not upload any to my MySQL. I checkt that all premission is on.

Error code:

Ihre Datenbank wird von der Version 0 auf die Version 26 aktualisiert:
Datenbank wird gesperrt...
Setting SQL_MODE...

Update v0 --> v1...
Step: CREATE TABLE IF NOT EXISTS `internal` (
`keyName` char(30) CHARACTER SET ascii NOT NULL,
`keyValue` varchar(255) COLLATE utf8_unicode_ci DEFAULT NULL,
UNIQUE KEY `keyName` (`keyName`)
) ENGINE=InnoDB DEFAULT CHARSET=utf8 COLLATE=utf8_unicode_ci; ...ERROR!
Error message: There is no active transaction

Setting SQL_MODE...
Releasing database...
ABORT: Could not perform update!

Entry "dbVersion" does not exists in table "internal"! 
------

And before this error comes, i get in to this meny:

Property | Value
-- | --
Current version: | 0
Needed version: | 26
Database size | 0 MB

----

Thanks for help!. :)

BjornOgren, 2021-02-19:
Okej! I solved it faster then i thought. 
I had php 8, and downgrade to 7, and it solved the problem.

jbtronics, 2022-03-02:
This was a incompatibility bug with PHP 8 which should be fixed now in version 0.5.11