Part-DB/Part-DB-legacy #107: Bilderupload Bauteile Error: 	Unbehandelte Exception

tx-rx-de, 2022-06-17:
TypeError:explode() expects parameter 2 to be string, null given [/mnt/web502/c2/96/510731996/htdocs/tx-rx/elebala/inc/lib.php, 269]

Wenn ich mit dem Browser eine Lokale Datei hochladen möchte *png/*jpg,
wird unter Dateiname nichts angezeigt.
Rechte für den Data-Ordner sind vergeben.

[image]

Stefan-Xp, 2022-12-17:
Hallo,

es ist zwar vielleicht kein guter Fix, aber ich habe in /inc/lib.php

Zeile 296:
`//Dont allow upload of files with potentially dangerous extension
    if (in_array(getExtensionFromFileName($file_array['name']), $unsafe_extensions)
        || in_array(getExtensionFromFileName($destination_filename), $unsafe_extensions))
    {
        throw new \Exception(_("Es ist nicht erlaubt PHP Dateien hochzuladen!"));
    }`

zu 

`    //Dont allow upload of files with potentially dangerous extension
    if (in_array(getExtensionFromFileName($file_array['name']), $unsafe_extensions))
    {
        throw new \Exception(_("Es ist nicht erlaubt PHP Dateien hochzuladen!"));
    }`

geändert, ich weiss zwar nicht genau warum oder ob es das unsicherer macht, aber zumindest funktioniert es so bei mir.

jbtronics, 2022-12-18:
Das Problem sollte jetzt in Part-DB 0.5.13 behoben sein

closed 2022-12-18 by jbtronics