Part-DB/Part-DB-legacy #110: Bauteile aus alter Datenbank importieren

rieders, 2022-08-25:
Hallo
Ich benutze schon einige Jahre Part-DB.
Nach längerer Pause habe ich versucht ein DB Update von v19 auf die aktuelle Version zu machen.
Leider bringt Part-DB dann immer eine Fehlermeldung.
Also habe ich Part-DB neu installiert und wollte die Bauteile dann ich die Tabelle Part einfügen.
Leider ist die Tabellenstruktur geändert.

Wie kann ich die Bauteile von der Alten DB in die Neue einfügen?

Grüße aus Thüringen

jbtronics, 2022-12-18:
Wie sieht die Fehlermeldung bei der Datenbankmigration denn genau aus?

rieders, 2022-12-19:
Hallo. Ich habe ja fast nicht mehr mit einer Antwort gerechnet. Ich muss erstmal noch mal nachschauen . Ich hatte es dann aufgegeben und versucht alles neu an zu legen. Ich müsste jetzt eine neue Datenbank anlegen um die jetzige nicht zu zerstören. Grüße André Von meinem iPhone gesendet