Part-DB/Part-DB-legacy #120: USB Webcam als Scanner

rieders, 2023-06-25:
Hallo 
Ich habe part-db unter Proxmox auf einer Virtuellen Umgebung erstellt.
Die angeschlossene Webcam habe ich auch durchgereicht.
Leider wird die Webcam von part-db nicht erkannt.
Unter Linux schon Bus 002 Device 002: ID 046d:0825 Logitech, Inc. Webcam C270.

Wie kann ich die Webcam richtig einbinden ?

Grüße aus Thüringen
André

jbtronics, 2023-06-25:
Es wird nicht die Webcam des Servers benutzt, sondern des verwendeten Endgeräts. Also des Smartphones, Computers, etc...

rieders, 2023-06-25:
Ok. Am endgerät habe ich auch eine Webcam.
Wie kann ich diese benutzen ?

rieders, 2023-06-25:
Vielen Dank für die schnelle Antwort.
Wie kann ich ein Update von 1.3- auf die aktuelle Version machen um nicht alles neu einzurichten.

Grüße

jbtronics, 2023-06-28:
Wenn du von der alten Version in diesem Repo auf die moderne Part-DB Version migrieren möchtest, findest du hier die Infos: https://docs.part-db.de/upgrade_legacy.html

Das mit dem Barcode scannen per Webcam funktioniert nur in der neuen Version. Die alten (legacy) Part-DB Version unterstützt nur Handscanner die direkt den Barcode in Text umwandeln.

rieders, 2023-06-29:
Vielen Dank für die Hilfe.
Ich habe die Version 1.5 in einem neuen Ordner installiert. 
Wie kann ich die Daten von der alten Version in die neue übertragen? 
Ich benutze nur sql.

Der Scanner geht leider auch nicht in der 1.5 er Version. 

Ich bin mal auf die german Demo gegangen,  da funktioniert es. 

Vielen Dank für die Hilfe. 

Grüße

rieders, 2023-07-07:
Hallo 
Ich habe jetzt mal die partDB legacy installiert. Wenn es dort die Möglichkeit Barcodes zu erstellen und zu scannen gibt würde mir die Version reichen. Ich habe auch noch die Server-Version laufen. Ich glaube das Problem mit dem Scanner ist dort das ich die IP Adresse nicht als vertrauenswürdige Webseite unter Chrome hin bekomme. Ich würde mir für die legacy Version mir auch ein Kostengünstigen Scanner kaufen wenn es mit dem Server nicht klappt.

Vielen Dank für die Hilfe

André