Part-DB/Part-DB-legacy #122: Cannot login with either admin or user pass

ObligedTester, 2024-01-13:
Hello,

Started up my old install of partdb legacy today to begin the migration work to another server and the latest version.

I encountered a strange issue, im not able to login to neither of my user accout or my admin account.
I checked the users directly in the phpadmin interface, and i am using the correct login information.
Is there a lockout timer for to many attempts? I might have used the incorrect info a couple of time before checking it in phpmyadmin

jbtronics, 2024-01-13:
The permissions system was changed and therefore all user permissions were reset.
To prevent problems, all users except the admin account with ID: 1.

Check which user has this ID by running: `php bin/console partdb:users:list`

It should be possible to login with the old password. However it that does not work you can reset the password with `php bin/console partdb:users:set-password [username]`

Afterwards login as the admin user, check the permissions of the other user and remove the checkbox at `User disabled` to allow them to login again

ObligedTester, 2024-01-13:
Thanks for your reply!
I will try this and get back to you.

However, i have not updated anything yet. This is my old install that worked before being offline for quite some time.
So its still legacy version. Any clues why it wont accept the password that is stored when checking the database?

jbtronics, 2024-01-13:
Does any error messages show up in Part-DB or in the browser console?

If you have accessed Part-DB in the past via https and now using just http, the browser might not accept a new cookie.

ObligedTester, 2024-01-13:
The only error is the usual "incorrect username / password" on top of the page

ObligedTester, 2024-01-14:
tried the "php bin/console partdb:users:list" now, im guessing straight in the terminal?
I get this:
Could not open input file: bin/console
Are these instructions for a legacy installation btw?

I once again checked the password inside the "users" table inside of phpmyadmin under the partdb database. And the names and passwords are correct. I just dont understand why it gives me "Warning! The username or password was wrong! " when i try any of them. 
It still works to browse around and check on parts, i guess as anonymous. But i cannot login, so im essentially locked out of the user interface.

I really need to sort this before moving to a new server and version. Its just so weird..

closed 2024-01-14 by ObligedTester